antirez/dump1090 #66: Feature Request: Aircraft type

Marsfan, 2015-01-22:
Add identification of aircraft type.

yuvadm, 2015-01-22:
@Marsfan Are you sure this is part of the ADS-B spec? I'm not sure at all.

Marsfan, 2015-01-22:
I'm not sure. I just know that another application I use (adsb#/Adsbscope) can tell me what plane type it is (b763, a321, etc).

hhrhhr, 2015-01-22:
Adsbscope use a 3rd-party db (adsbscope\extra\icao24plus.txt)

marutan, 2015-06-05:
I thought I'd drop a line to say that I've implemented this, reading the icao24plus.txt file from adsb scope. I just need to tidy the patch up slightly (and then create a repo on github to pull from)

SeanMaday, 2015-06-09:
Is it possible to have this field passed to data.json?

marutan, 2015-06-09:
Yes, infact I only output this on the JSON, I didn't want to change the format of anything where dump1090 was acting as 'headless'.

I realised only later that I'd done this on the MalcommRobb fork of dump1090, you can see the pull request and a screenshot over here.

https://github.com/MalcolmRobb/dump1090/pull/70

quentin9696, 2016-03-24:
Hello, were can I find an updated file of icao24.txt ? 

Is this file is up to date http://www.tech-software.net/RTL1090/adsb_scope/pc_software/adsbscope/26/extra/icao24plus.txt ? 

Regards,