antirez/dump1090 #67: Make Error

scuba323, 2015-02-03:
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr` -c dump1090.c
Package librtlsdr was not found in the pkg-config search path.
Perhaps you should add the directory containing `librtlsdr.pc'
to the PKG_CONFIG_PATH environment variable
No package 'librtlsdr' found
dump1090.c:46:21: fatal error: rtl-sdr.h: No such file or directory
compilation terminated.
make: *** [dump1090.o] Error 1

I have RTL_SDR installed

yuvadm, 2015-02-03:
> I have RTLSDR installed

Your make attempt begs to differ. Did you properly install it with a package manager?

scuba323, 2015-02-04:
Yes using gqrx's Launchpad PPA. I have Elementary OS.

yuvadm, 2015-02-04:
This is an OS specific problem - either a bad PPA, or some local problem on your machine - and I'm not familiar with Elementary OS so I can't help much.

Generally speaking, if you have librtlsdr properly installed, you should be seeing something like this:

```bash
$ pkg-config --cflags librtlsdr
-I/usr/include/ -I/usr/include/libusb-1.0
```

scuba323, 2015-02-04:
Package librtlsdr was not found in the pkg-config search path.
Perhaps you should add the directory containing `librtlsdr.pc'
to the PKG_CONFIG_PATH environment variable
No package 'librtlsdr' found

I don't get it. when I try to apt-get, it says latest version

yuvadm, 2015-02-04:
Please close this ticket, this is not a dump1090 issue.

closed 2015-02-04 by scuba323

JLospinoso, 2018-01-22:
@scuba323 You might try installing from source, e.g.:

```
git clone git://git.osmocom.org/rtl-sdr.git
mkdir build
cd build
cmake ..
make
make install
```

rfilmyer, 2019-05-22:
Hint: If you're on a Debian system, you should check if you installed `librtlsdr-dev`. Not the rtl_sdr tools, not `librtlsdr0`, but `librtlsdr-dev`.