antirez/dump1090 #75: Bad use of LDFLAGS in commit 36cec38

xmw, 2015-03-10:
See comments on https://github.com/antirez/dump1090/commit/36cec38231eada02386dabec85355eba195352c5 for details

    --- dump1090-9999/Makefile
    +++ dump1090-9999/Makefile
    @@ -1,5 +1,5 @@
     CFLAGS?=-O2 -g -Wall -W `pkg-config --cflags librtlsdr`
    -LDFLAGS?=`pkg-config --libs librtlsdr` -lpthread -lm
    +LDLIBS+=$(shell pkg-config --libs librtlsdr) -lpthread -lm
     CC?=gcc
     PROGNAME=dump1090
     
    @@ -9,7 +9,7 @@
            $(CC) $(CFLAGS) -c $<
     
     dump1090: dump1090.o anet.o
    -       $(CC) -g -o dump1090 dump1090.o anet.o $(LDFLAGS)
    +       $(CC) -g -o dump1090 dump1090.o anet.o $(LDFLAGS) $(LDLIBS)
     
     clean:
            rm -f *.o dump1090```

yuvadm, 2015-03-10:
@xmw Thanks for notifying of breakage. Please open a proper PR that I can merge.

xmw, 2015-03-10:
https://github.com/antirez/dump1090/pull/76

closed 2015-03-10 by yuvadm