antirez/dump1090 #83: Interactive shows flights not visible on the net HTML page

ricsim, 2015-05-27:
I use this commandline:
./dump1090 --metric --enable-agc --net --net-http-port 8080 --interactive --aggressive
(need to cd to dir otherwise html file will not be found ... see other bug report here)

There are many flights with known position and track. (see below interactive screenshot)
But it shows currently 0 planes on the HTTP GUI.
Why?!?

Hex    Flight   Altitude  Speed   Lat       Lon       Track  Messages Seen  .
--------------------------------------------------------------------------------
2000f7          0         0       0.000     0.000     0     1         15 sec
3c646b CFG6AM   11560     833     0.000     0.000     320   64        11 sec
4404d5          0         0       0.000     0.000     0     1         41 sec
8963ce          11270     0       0.000     0.000     0     3         55 sec
3c6490 DLH5AC   10029     842     48.210    15.945    91    612       0 sec
3c6593 DLH8NW   11270     861     47.556    15.618    114   394       0 sec
471ea7          10966     814     0.000     0.000     309   8         1 sec
506c07          9747      0       0.000     0.000     0     1147      0 sec
3c6561 DLH639   10357     807     47.837    15.710    291   1638      4 sec
4a8188 SVF820   0         0       0.000     0.000     0     565       0 sec
3c544c DLH8CR   10661     872     47.598    15.862    108   674       22 sec
405ee1 BAW697   3762      655     48.252    16.126    334   2073      0 sec
506c1a          9443      0       0.000     0.000     0     1299      1 sec
451e84 VIM371   11567     829     48.245    15.700    294   2574      25 sec

JamesHagerman, 2015-06-07:
I was having this problem as well on OS X. The issue is related to the rather horrible HTTP Request parsing code in this project.

Basically, if you've somehow got cookies attached to http://localhost:8080/, they will be sent in the request to the dump1090 parser. This doesn't match the assumptions the parsing code is expecting.

The easiest solutions are using a private or incognito browsing session... or just try clearing your cookies.

It seems to work after that!

ricsim, 2015-06-07:
No cookies set.
And I _DO_ see planes, just much less than there are on the interactive view (eg. I see planes in interactive with coordinates, and track that do not show up in the web view)