antirez/dump1090 #85: Uninitialized pointer in dump1090.c

JerryHRoss, 2015-06-20:
I have found an uninitialized pointer in the dump1090.c code that prevents bringing up two sdr devices simultaneously on windows.

Here is the code from dump1090.c with the fix in place:

int verbose_device_search(char *s)
{
	int i, device_count, device, offset;
  char temp[16];
  char *s2 = temp;
	char vendor[256], product[256], serial[256];

I now have two devices running with separate images of dump1090 running and can start/stop them in any combination without them interfering with each other in the process.  A third device will be tested this afternoon.

Jerry Ross

yuvadm, 2015-06-20:
Thanks @JerryHRoss, can you please submit a proper pull request so we can merge this fix?

JerryHRoss, 2015-06-20:
This is actually a bug in the MalcolmRobb fork of dump1090.  Having retired from the industry before git became popular I am at a loss to what a "proper pull" would be.  Let me know if I still need to do the "pull".

JerryHRoss, 2015-06-20:
Okay,  Went to the MalcomRobb fork, pulled it, entered the fix, and committed it.

JerryHRoss, 2015-06-20:
Still learning.  I don't have permission to sync to that repository.  What do I need to do for this fix to get merged with the master?

yuvadm, 2016-01-04:
This bug seems to apply only to MalcomRobb's fork, closing.

closed 2016-01-04 by yuvadm