antirez/dump1090 #89: No device found

belisards, 2015-10-06:
rad1o (hackRF) does not work (without sudo too) with dump1090, any ideia why?

abitporu@hpnote:~/rad1o/dump1090$ sudo hackrf_info 
Found HackRF board.
Board ID Number: 3 (Unknown Board ID)
Firmware Version: git-dfbbbfb
Part ID Number: 0xa0000a30 0x00624747
Serial Number: 0x00000000 0x00000000 0xa83065d9 0x213f5127
abitporu@hpnote:~/rad1o/dump1090$ sudo ./dump1090 
No supported RTLSDR devices found.

yuvadm, 2015-10-06:
HackRF isn't an RTLSDR device.

belisards, 2015-10-06:
ooops, sorry for this dumb question, but is there a way to run dump1090 with hackrf?

yuvadm, 2015-10-06:
Use any hackrf program you have to pipe the data to stdin:

    hackrf_whatever -f 1090000000 | dump1090 --ifile -

closed 2015-10-06 by yuvadm