antirez/dump1090 #102: dump1090 port 30003 FlightID seems space padded

vk4tec, 2016-04-02:
SBS Port 30003 from dump1090 FlightID seems space padded in output

Could give the impression aircraft is sending space padded FlightID

This is not allowed in our country 

Andrew Rich 

Radar Support - Australia

Example 

,FD467   ,,,,,,,,,,,0 ( space padded ) FlightID

Should if correct read 

,FD467,,,,,,,,,,,0

MSG,1,111,11111,7C19C4,111111,2016/04/02,22:29:54.173,2016/04/02,22:29:54.143,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:29:59.014,2016/04/02,22:29:58.990,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:13.905,2016/04/02,22:30:13.869,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:23.642,2016/04/02,22:30:23.629,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:28.714,2016/04/02,22:30:28.678,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:38.756,2016/04/02,22:30:38.707,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7501E8,111111,2016/04/02,22:30:44.720,2016/04/02,22:30:44.670,XAX207  ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:29:54.173,2016/04/02,22:29:54.143,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:29:59.014,2016/04/02,22:29:58.990,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:13.905,2016/04/02,22:30:13.869,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:23.642,2016/04/02,22:30:23.629,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:28.714,2016/04/02,22:30:28.678,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:30:38.756,2016/04/02,22:30:38.707,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:31:09.121,2016/04/02,22:31:09.108,FD467   ,,,,,,,,,,,0
MSG,1,111,11111,7C19C4,111111,2016/04/02,22:31:19.138,2016/04/02,22:31:19.134,FD467   ,,,,,,,,,,,0

yuvadm, 2016-04-02:
@vk4tec are you sure this isn't the MalcomRobb fork you're referring to? I don't think this version of dump1090 provides such extensive output.

vk4tec, 2016-09-01:
Annex 10 says spaces are ok 

Andrew