antirez/dump1090 #103: Metric system fail

inetsec, 2016-04-25:
The metric mode in the browser isn't working properly...
Here is the code which need to change...
2156 a->altitude should be altitude (or the metric system won't be displayed in the browser)
2157 a->speed should be speed  (or the metric system won't be displayed in the browser)

Changed code:
        if (a->lat != 0 && a->lon != 0) {
            l = snprintf(p,buflen,
                "{\"hex\":\"%s\", \"flight\":\"%s\", \"lat\":%f, "
                "\"lon\":%f, \"altitude\":%d, \"track\":%d, "
                "\"speed\":%d},\n",
                a->hexaddr, a->flight, a->lat, a->lon, altitude, a->track,
                speed);

And the browser gmap.html need also a change to support the metric system (but I don't know how I would change this in a way that it supports the feet and meters ...)

Besides this few errors is this program is awesome ;)
Thank you for your nice work :)

lops777, 2016-05-09:
For the web browser I just changed the file "public_html/config.js" to  have the entry:

    Metric = true; // true or false

Since then I could see in the web browser metric units...
Greetings

antirez, 2026-02-15:
Fixed.

closed 2026-02-15 by antirez