antirez/dump1090 #116: No planes on console or plotter.

jdoscher, 2017-04-13:
I have a new Raspberry Pi 3 and a Fitipower FC0012 tuner with a stock antenna. I've tried several different builds of Dump1090, but they all exhibit the same behavior- when I start the app, all I get is:

> Found 1 device(s): 0: Realtek, RTL2838UHIDIR, SN: 00000001 (currently selected)
> Found Fitipower FC0012 tuner
> Max available gain is: 19.20
> Setting gain to: 19.20
> [FC0012] no valid PLL combination found for 1090000000 Hz!
> Exact sample rate is: 2000000.052982 Hz
> Gain reported by device: 19.20

I've tried using a powered USB hub, and even placed the antenna in a flight path for the local airport, and still have never picked up a flight. Any ideas? Thanks!

yuvadm, 2017-04-13:
This doesn't seem like a dump1090 issue. Some suggestions:

- Test with a different dongle (ideally with a different tuner)
- Try a different antenna
- Make sure you are near an ADS-B transmitting aircraft, being within line-of-sight from a large commercial airliner should deifnitely be enough
- Check with signal analysis software (such as gqrx) whether you are actually receiving anything on 1090Mhz

If you conclude that dump1090 is buggy, please reopen this issue with more details.

closed 2017-04-13 by yuvadm

jdoscher, 2017-04-13:
Thank you for confirming this is not a dump1090 issue.