antirez/dump1090 #124: Position and velocity messages getting repeated 2 to 3 times

sekarchandran, 2017-09-28:
Hi

Thanks to creators of this wonderful tool ..which helped me lot in my work.

I am developing a tool which will compare my Radar data  with the ADSB (from dump tool) data for bias calculation.

I am using the "NooElec NESDR Mini 2 SDR & DVB-T USB Stick "   and dump1090 .

I have following problems

1.  Many times, Messages  are getting repeated with in short time.. 100 ms ( maximum three times)  same values ..only small time difference...?

I tried following methods ...but  not able to solve the problem
       a)  changed the buffer size and number of buffers in rtlsdr_read_async function..
       b)  changed the receive using  "rtlsdr_read_sync " 
       c)  removed mutex and signals ...instead i used POSIX  message queue bet receive and process 
threads...

still data is repeating some times .

Is it the bug in rtlsdr library which makes repeated messages OR  ADSB system itself broadcast messages like this ...?

Kindly help me.

Thanks and Regards
K.Chandrasekaran

yuvadm, 2017-09-29:
It's definitely possible the aircraft broadcast the same message several times in response to different interrogations.

sekarchandran, 2017-09-29:
ADSB messages( position -lat,long,altitude, velocity - speed,heading, call sign..)  are broadcasted  without any interrogation at regular intervals ( one second once) ....correct me if i am wrong .. 

I just added some code to display the messages in console continually ....there i noticed that  both position and velocity messages are repeated with in some milliseconds interval...same values only time is the different.

I verified the code thoroughly with various ways...in my understanding ..these messages are coming in different times ( different reads )
have you ever noticed that the position and velocity messages repeated ..?

yuvadm, 2017-09-29:
As far as I understand ADSB messages can either be periodical (as is the case with extended squitter) or respond to secondary radar interrogations.

Regardless, unless it can be shown that there truly is a bug that causes erroneous repeated messages I'll have to close this issue.

closed 2017-09-29 by yuvadm

sekarchandran, 2017-09-29:
as of now  i am not having the dongle with me ... i will come back  with the log data after getting the dongle ..probably two to three days ..