antirez/dump1090 #126: No flights are shown on the list

vlabots, 2017-12-10:
Hi, i've tried many times to parse the flights with an R820T tuner, but every time `./dump1090 --interactive --net` is giving me the empty list. 

That's the result, that is showing after running a `./dump1090` command:
`Found 1 device(s):`
`0: Realtek, RTL2838UHIDIR, SN: 00000001 (currently selected)`
`Found Rafael Micro R820T tuner`
`Max available gain is: 49.60`
`Setting gain to: 49.60`
`Exact sample rate is: 2000000.052982 Hz`
`Gain reported by device: 49.60`

What's the problem can be? What am I doing wrong?

fantuz, 2017-12-10:
well, it could be that your antenna is "mute" on 1090mhz (i.e. not a
scanner antenna, not a wideband one...)

have you tried with "croatia microwave" or specific ADSB antenna? anyway,
gain at max is OK as signals are not so strong (almost 1.1 gigahertz)
though no obstacles and clear path (try to go outside on a terrace) my

try using a generic 433 antenna and see if packets get through

regards,
max

vlabots, 2017-12-10:
@fantuz Hi, Max.
That's the screen with my antenna's description: 
http://joxi.ru/l2ZLexDU8GzNVr

As I understood, this kind of antenna should work fine on these frequencies...

Sherdil675, 2022-02-17:
@vlabots dear were you able to solve this issue? Using the similar antenna and kerberos sdr I am able to receive a number of aircrafts using the rtl_adsb but with dump1090 it stops at the Gain point. It is stuck there though I waited for more than an hour but no success. A day before it was working but now after trying for a number of times the issue is still there and it is similar to your issue.

rsavignon, 2022-10-15:
i Either. How can solve the problem ???

> @vlabots dear were you able to solve this issue? Using the similar antenna and kerberos sdr I am able to receive a number of aircrafts using the rtl_adsb but with dump1090 it stops at the Gain [...]

FraOre, 2023-07-05:
Hi! I have a similar @vlabots situation:
if I run dump1090 I get this message (and no flights):

`
Found 1 device(s):
0: Realtek, RTL2838UHIDIR, SN: 00000001 (currently selected)
Found Fitipower FC0012 tuner
Max available gain is: 19.20
Setting gain to: 19.20
[FC0012] no valid PLL combination found for 1090000000 Hz!
Exact sample rate is: 2000000.052982 Hz
Gain reported by device: 19.20
`

Ok I understood (maybe...) that my tuner (Fitipower FC0012) doesn't support the standard flight's message frequency (1.09Ghz) as explained here: https://www.rtl-sdr.com/about-rtl-sdr (do I solve It buying another hardware with a R820T chip?!)...

But at the same time, if I just run rtl_adsb a lot of messages scroll on my terminal:

`
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Tuner gain set to automatic.
[FC0012] no valid PLL combination found for 1090000000 Hz!
WARNING: Failed to set center freq.
Exact sample rate is: 2000000.052982 Hz
Sampling at 2000000 S/s.
*9e4e58174f0b356f7ee86dea2f12;
*adb493641bb4a451285472fe3dc4;
*84d49acb9ab7a056f795a6a35cde;
*944a119fcca9e4eca9269d3ddbe6;
*e1e5d0f755d976ad8a7f567898eb;
*b2a7b908b7d515f8d4d7136dabd2;
*aa62b32d0e8e75456e8cb9057226;
*cc52b55eb92edfa4cf19fc671ffc;
*b969e712ef06c54291faca91750c;
*9d6fd33e0138c660ca6e3dc8af89;
and a lot more...
`

so now what I don't understand is if all that messages (starting with an asterix) are flights' messages (and therefore potentially decodable) or what?
I don't understand if the problem is hardware (you're not reading the flights' data but just garbage, so buy a better dongle!!!) or if the messages are in fact valid and the problem is software...

Really thank you very much to whoever will help me clear up my ideas.

Francesco