antirez/dump1090 #136: Web interface needs an update

FelixTremblay, 2018-07-16:
Google maps seems to have changed some things with its api. When I load the page, I am greeted with a popup saying : "this page can't load google maps correctly". After dismissing the alert, the map shown is really dark and has a bunch of "for development only" watermarks.

jay-to-the-dee, 2018-09-06:
Can confirm this issue. Unfortunately solution @HarryMJensen didn't work for me :(

jay-to-the-dee, 2018-09-06:
It appears their needs to be an API key for this to work now. Worth mentioning how to do this in the README doc?

HNJAMeindersma, 2018-11-03:
About a year ago Google did some things to their  Maps API. Other branches of dump1090 and I believe even this one has moves to the OpenStreetMaps API since then.

danstreeter, 2018-12-27:
As above, I've requested a PR that fixes this issue - I've done it in a forked repo which my codebase derives from which is also further ahead of this one: although the changes made are very small and are easily transferable as required.

PHPoenX, 2019-06-30:
> OpenStreetMaps

It is wrong shows a lot of locations

antirez, 2020-02-03:
Hi, I ported the code to open street map. Closing.

closed 2020-02-03 by antirez