antirez/dump1090 #142: Build issue (pkg-config installed, librtlsdr-dev installed)

AnnoyingTechnology, 2019-03-23:
Hello,

here it is : 

```bash
cc -g -o dump1090 dump1090.o anet.o  -L -lrtlsdr -lpthread -lm
/usr/bin/ld: dump1090.o: in function `modesInitRTLSDR':
/home/julien/Downloads/dump1090-master/dump1090.c:340: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:348: undefined reference to `rtlsdr_get_device_usb_strings'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:353: undefined reference to `rtlsdr_open'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:360: undefined reference to `rtlsdr_set_tuner_gain_mode'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:372: undefined reference to `rtlsdr_set_tuner_gain'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:377: undefined reference to `rtlsdr_set_freq_correction'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:379: undefined reference to `rtlsdr_set_center_freq'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:380: undefined reference to `rtlsdr_set_sample_rate'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:381: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:383: undefined reference to `rtlsdr_get_tuner_gain'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:378: undefined reference to `rtlsdr_set_agc_mode'
/usr/bin/ld: /home/julien/Downloads/dump1090-master/dump1090.c:368: undefined reference to `rtlsdr_get_tuner_gains'
/usr/bin/ld: dump1090.o: in function `readerThreadEntryPoint':
/home/julien/Downloads/dump1090-master/dump1090.c:460: undefined reference to `rtlsdr_read_async'
/usr/bin/ld: dump1090.o: in function `main':
/home/julien/Downloads/dump1090-master/dump1090.c:2632: undefined reference to `rtlsdr_close'
collect2: error: ld returned 1 exit status
make: *** [Makefile:12: dump1090] Error 1

```

```bash
$ apt search librtlsdr
Sorting... Done
Full Text Search... Done
gr-osmosdr/testing,now 0.1.4-14+b10 amd64 [installed,automatic]
  Gnuradio blocks from the OsmoSDR project

librtlsdr-dev/testing,now 0.6-1 amd64 [installed]
  Software defined radio receiver for Realtek RTL2832U (development)

librtlsdr0/testing,now 0.6-1 amd64 [installed]
  Software defined radio receiver for Realtek RTL2832U (library)

```

Thanks for your help

CJHackerz, 2019-03-24:
I am also getting same problem

hopkinskong, 2019-04-26:
The issue is `pkg-config` generates an empty `-L`.
In my case, do the following:
```
pkg-config --libs librtlsdr libhackrf libairspy soxr
```
1. Copy the output of above command. In my case, it was `-L -lrtlsdr -lhackrf -lairspy -lsoxr`.
2. Modify the `LDLIBS` variable in the `Makefile`. Substitute the `pkg-config` expansion (`$(shell pkg-config ...)`) with above output. Make sure you omit the initial `-L`, so mine should be `-lrtlsdr -lhackrf -lairspy -lsoxr`.
3. Run `make`
4. Have fun :)

filippog, 2019-08-04:
If the build failure is happening on Debian-based system then it is likely [bug #925916]( https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=925916) (fixed in testing and unstable). The tl;dr is to have this at the top of `/usr/lib/x86_64-linux-gnu/pkgconfig/librtlsdr.pc` instead of empty values:

```
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include
```

DakotaNelson, 2020-05-21:
Building on what @filippog said above, on a raspi run `pkg-config --libs librtlsdr --debug`, which, for me, outputs the location of `librtlsdr.pc` as `/usr/lib/arm-linux-gnueabihf/pkgconfig/librtlsdr.pc` - editing it then fixed the issue.

Dygear, 2020-05-23:
I went from this error https://github.com/antirez/dump1090/issues/108 to this error here. Even with doing `pkg-config --libs librtlsdr --debug` I'm still getting the same error. What do I need to edit to make this work?

m1zar, 2020-07-25:
Was using the MalcolRobb/dump1090 version but got this error, finally found that using the version at https://github.com/joergsteinkamp/dump1090, compiles fine. And actually working better than ever! :o)

Dygear, 2020-11-04:
Had to build this again as I moved over to a new system with a new arch. Was surprised that this issue was still open and outstanding, and even more that I had ready commented and fixed the problem. Thanks past me.

iXmerof, 2022-03-21:
In 2022 for rpi zero the issue was still present, the @[hopkinskong](https://github.com/hopkinskong) response helped (for lrtlsdr only)

closed 2024-02-07 by yuvadm