antirez/dump1090 #147: Google Maps needs a key and billing enabled

schroeshirecat, 2019-11-17:
Google Maps now no longer works if you don't have a valid key, which requires an account with Google with billing enabled. I am going to try to adapt this to OpenStreetMaps, but that's going to take a while.
The rest of the program works flawless out of the box. Thanks for the good work.

gch1p, 2019-12-29:
If anybody wants to fix the map for themself:

1. go through a process of creating a billing account and receiving an API Key;
2. open `gmap.html` file, find this:
`<script type="text/javascript"
      src="https://maps.googleapis.com/maps/api/js?sensor=true">
` and append `&key=YOUR_API_KEY` to the url.
3. rename `Map` variable to `map`:
    - Change `Map=null;` to `map=null;`
    - `map: Map` to `map: map`
    - `Map = new google.maps.Map` to `map = new google.maps.Map`

Then it should work.

antirez, 2020-02-03:
Hi, I ported the code to open street map. No longer needed. Thanks.

closed 2020-02-03 by antirez

schroeshirecat, 2020-02-27:
Cool, thank you. In the mean-time I've made my own version https://github.com/ofehr-media/dump1090

DC7IA, 2021-11-06:
> going to try to adapt this to OpenStreetMaps

suggestion: offline maps

Because sometimes you don't have an internet connection.

Something like the option to use a very basic map would be useful then. :D

r3k2, 2023-12-13:
@antirez I just got the git version from the arch Linux AUR and still used the bad Google Maps instead of good OpenStreetMap ... I thought you said you ported the code?