antirez/dump1090 #149: How can I get the amplitude map of IQ data in debug mode?

1benpaodewoniu-wp, 2020-01-09:
Hello there,
Your project is very good. I am very happy to follow your open source project and follow your step by step instructions to successfully track and obtain the aircraft information. The problem now is how to display the IQ data. I still do not understand how to do it The Debug.html file loaded directly by the browser on the terminal cannot be displayed. What should I do? Where is the original IQ data stored and how is it obtained? thank you very much!

sdradsb, 2020-01-30:
I am also facing the same issue. I would like to know from where I can get the I/Q data or where it is being dumped. Did you by any chance figure it out? I would really appreciate if anyone could help me with this.

dgoadby, 2020-01-31:
I took a look at the code in dump1090.c . The SDR reads are done in a background thread using a callback "rtlsdrCallback" which is set in "readerThreadEntryPoint". The easiest place to get I/Q data would be the function void "computeMagnitudeVector(void)" as this takes the I/Q data from Modes.data to calculate a magnitude vector.  You could use a pipe or FIFO to send the I/Q data from here to another task.

sdradsb, 2020-02-10:
Thank you for the reply. I am trying to collect the raw IQ for each ADS-B signal. Right now I am getting the output.bin which has all the IQ samples. Is there anyway I could differentiate  an IQ for each ADS-B signal. I would really appreciate your help. Thank you

1benpaodewoniu-wp, 2020-03-05:
> 
> 
> Thank you for the reply. I am trying to collect the raw IQ for each ADS-B signal. Right now I am getting the output.bin which has all the IQ samples. Is there anyway I could differentiate an IQ for each ADS-B signal. I would really appreciate your help. Thank you

Hello, do you know how to store IQ raw data? In the command I got the decoding information of the ADS-B signal, but I still don't know how to store the original data. If anyone can help me, I would appreciate it, thank you!

1benpaodewoniu-wp, 2020-03-06:
Hello! Did you get the original IQ data output.bin file through the following command?
rtl-sdr -f1090000000-s 2000000-g50 output.bin
How can this file be found on the computer? Or does it only show results on the terminal?
I would be especially grateful if someone could help me!

totot077, 2021-01-03:
وءىعء

totot077, 2021-01-03:
s do