antirez/dump1090 #158: Flights not detected after certain altitude

saikumarelab, 2021-02-11:
Hi I have installed my adsb antenna on my roof top and its a two storey building with 10ft height. I cannot able to detect the flights which are landing in airstrip since I'm located close to it . I can able to detect the flights at <1250ft. How do we correct this to receive flights on ground.

dgoadby, 2021-02-11:
Which ADSB antenna are you using? The problem is probably due to the polar response of the aerial. Aerials are always a compromise and, at these frequencies, proximity to buildings and cabling can also have significant effects. Ideally you want a radiation patter that is a hemisphere so you would be able to see all of the sky right down to the horizon. A most of us do not have an airport close by then an antenna with a radiation pattern with more gain upwards (>25 degrees) is beneficial but planes on the ground will be much weaker. Also, at some airports I find the ADSB stops at point of landing.

saikumarelab, 2021-02-12:
Hey dgoadby, thanks for the reply. So it is understood that the antenna polarization and elevation angle are the key factors for signal reception. By the way do you recommend any specific antenna and the cable length to achieve the grounding aircrafts with low altitude. From our building the airport runway is around 7to 8kms air distance and In between we have building as obstructions.

dgoadby, 2021-02-13:
Well, my first antenna was a cheapie whip but that was a waste of time. It was a small whip with a magnetic base. In truth it did not even resonate at our frequency of interest (1090MHz). The second one was this one: https://shop.pimoroni.com/products/ads-b-1090-mhz-antenna-1-2m-8-5dbi. It works well for me but I do not need low level gain. I have recently ordered this one: https://www.banggood.com/1090MHZ-36DB-SMA-Active-ADS-B-PCB-Antenna-with-Biaser-Tee-Kit-p-1754082.html?cur_warehouse=CN&rmmds=search . It has a built in pre-amp and filter. The preamp is useful as the coax cable losses are significant at this frequency and the filter reduces SDR overload (cross modulation) by other strong UHF transmissions  .  In fact, the coax cable investment is probably more important than the antenna as 20 ft of poor cable will totally negate your antenna gain. What cable are you using?

closed 2026-02-15 by antirez