antirez/dump1090 #165: Not Compiling

tomcass240, 2021-09-27:
Getting the following error when typing make as per the instructions.

gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o `pkg-config --libs librtlsdr` -lpthread -lm 
/usr/bin/ld: interactive.o:/home/z/git/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/z/git/dump1090/dump1090.h:373: first defined here
/usr/bin/ld: interactive.o:/home/z/git/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/z/git/dump1090/dump1090.h:238: first defined here
/usr/bin/ld: mode_ac.o:/home/z/git/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/z/git/dump1090/dump1090.h:373: first defined here
/usr/bin/ld: mode_ac.o:/home/z/git/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/z/git/dump1090/dump1090.h:238: first defined here
/usr/bin/ld: mode_s.o:/home/z/git/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/z/git/dump1090/dump1090.h:373: first defined here
/usr/bin/ld: mode_s.o:/home/z/git/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/z/git/dump1090/dump1090.h:238: first defined here
/usr/bin/ld: net_io.o:/home/z/git/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/z/git/dump1090/dump1090.h:373: first defined here
/usr/bin/ld: net_io.o:/home/z/git/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/z/git/dump1090/dump1090.h:238: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:24: dump1090] Error 1

Zorglu, 2021-12-30:
the same

hoopertsau, 2022-01-18:
change the Modes and tDF define in dump1090.h. just define the struct type, not the instance. 
`
struct stModes {
XXX
XXX
XXX
....
};
struct stDF {
XX
XXX
...
};
`

and declare  extern 
`
extern struct stModes Modes;
extern struct stDF tDF;
`

finnally,put the Modes and tDF instance define into dump1090.c and view1090.c

`struct stModes Modes;
struct stDF tDF;`

you can now make it.

jamesg31, 2022-01-22:
> change the Modes and tDF define in dump1090.h. just define the struct type, not the instance. `struct stModes { XXX XXX XXX .... }; struct stDF { XX XXX ... };`
>
> and declare extern `extern struct [...]

I am having the same issue on a raspberry pi 4. Can you go into more detail with these instructions? I do not understand.

akshitmehra, 2022-01-25:
I had the same issue, recently. You can follow the steps below:

1. Open `dump1090.h` file
2. Go to line 238 and change to just `};`
3. Go to line 241 and change to `struct stModes {`
4. Go to line 373 and change to just `};`
5. Open `dump1090.c` file
6. Insert a line below line 31 and paste `struct stModes Modes; struct stDF tDF;`
7. Open `view1090.c` file
8. Insert a line below line 31 and paste `struct stModes Modes; struct stDF tDF;`
9. Run `make -B`

It should work fine now.

AgentSmith0, 2022-01-26:
@akshitmehra Thank you so much for your help, but I am trying to compile this fork: https://github.com/MalcolmRobb/dump1090
But I get the following error messages:
```
gcc -g -o dump1090 dump1090.o anet.o interactive.o mode_ac.o mode_s.o net_io.o `pkg-config --libs librtlsdr` -lpthread -lm 
/usr/bin/ld: interactive.o:/home/pi/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
/usr/bin/ld: interactive.o:/home/pi/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
/usr/bin/ld: mode_ac.o:/home/pi/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
/usr/bin/ld: mode_s.o:/home/pi/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
/usr/bin/ld: mode_s.o:/home/pi/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
/usr/bin/ld: net_io.o:/home/pi/dump1090/dump1090.h:373: multiple definition of `Modes'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
/usr/bin/ld: net_io.o:/home/pi/dump1090/dump1090.h:238: multiple definition of `tDF'; dump1090.o:/home/pi/dump1090/dump1090.c:32: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:24: dump1090] Error 1
```
Do you know how I can fix these errors?
And maybe you can create a PR for the fix?

AgentSmith0, 2022-03-04:
@McCavity Thank you so much for your help!!! It works fine now.

daniyals48, 2022-09-08:
> I had the same issue, recently. You can follow the steps below:
>
>     1. Open `dump1090.h` file
>
>     2. Go to line 238 and change to just `};`
>
>     3. Go to line 241 and change to `struct stModes [...]

Hello Can you share both of the files because in dump1090.c files there are comments and no code.

daniyals48, 2022-09-08:
> Here are my files:
> dump1090.h:
>
> // dump1090, a Mode S messages decoder for RTLSDR devices.
> //

Thank you so much.
I was wondering if have any time and you can help me with my college final
years project?
Since you have a lot of knowledge and I am newbie

gx1400, 2022-11-15:
> 

This resolved for me as well on an rPi4 8GB

1. Open `dump1090.h` file
2. Go to line 238 and change to just `};`
3. Go to line 241 and change to `struct stModes {`
4. Go to line 373 and change to just `};`
4a. Insert a line below line 373 and paste extern struct stModes Modes; extern struct stDF tDF;

5. Open `dump1090.c` file
6. Insert a line below line 31 and paste `struct stModes Modes; struct stDF tDF;`

7. Open `view1090.c` file
8. Insert a line below line 31 and paste `struct stModes Modes; struct stDF tDF;`

9. Run `make -B`

daniyals48, 2022-11-20:
Thank you for your steps... I test these steps for learning but I found an
easy solution for this... I used dragon OS which has the newer version and
it supports this feature by default.
😊😊

closed 2024-02-07 by yuvadm

Enddle, 2024-02-11:
same error on rpi4, followed the solution

> change the Modes and tDF define in dump1090.h. just define the struct type, not the instance. `struct stModes { XXX XXX XXX .... }; struct stDF { XX XXX ... };`
>
> and declare extern `extern struct [...]

and it gives a new error as below

`~/dump1090 $ make -B
gcc -O2 -g -Wall -W `pkg-config --cflags librtlsdr`  -c dump1090.c
In file included from dump1090.c:31:
dump1090.h:375:37: error: unknown type name ‘stDF’
  375 | extern struct stModes Modes; extern stDF tDF;
      |                                     ^~~~
dump1090.c:32:35: error: conflicting types for ‘tDF’; have ‘struct stDF’
   32 | struct stModes Modes; struct stDF tDF;
      |                                   ^~~
dump1090.h:375:42: note: previous declaration of ‘tDF’ with type ‘int’
  375 | extern struct stModes Modes; extern stDF tDF;
      |                                          ^~~
......
/usr/include/aarch64-linux-gnu/sys/timeb.h:29:12: note: declared here
   29 | extern int ftime (struct timeb *__timebuf)
      |            ^~~~~
make: *** [Makefile:21: dump1090.o] Error 1
`

below line 373 of `dump1090.h` file, I changed the code inserted from above from
`extern struct stModes Modes; extern stDF tDF;`
to
`extern struct stModes Modes; // extern stDF tDF;`

it still gave warnings but it builds and works  ;)