antirez/dump1090 #175: Include file "rtl-sdr.h" not found

mok0, 2023-07-23:
When I clone the git repo, and type "make" I get an error from the compiler that include file "rtl-sdr.h" is not found.

Obviously there is a missing package, but I have not yet found out where it comes from.

mok0, 2023-07-23:
So, it seems on Debian derived systems you need to install  `librtlsdr-dev`.

Perhaps that should be made clear from the installation instructions that this package is a requirement for building `dump1090`.

alexmyczko, 2023-07-23:
@mok0 it's so obvious, you figured within 5 minutes. i'm sure others will do similar.

iwelch, 2023-10-23:
please mention what it depends on on the page anyway.  the cost of doing so is minimal.  if nothing else, it gets you not to have to look at issues.

yuvadm, 2024-02-07:
@iwelch feel free to open a PR clarfying the prerequisites.